grodansparadis/vscp #231: VSCP driver won't load

mamac, 2017-12-27:
Hi,

On 12.29.2.30 Magnesium

I have set the following driver in /etc/vscp/vscp.conf:
`        
`        <driver enable="true" >
`            <name>VSCP Level II SocketCAN Driver A</name>
`            <path>/srv/vscp/drivers/level2/vscpl2drv-socketcan.so</path>
`            <config>can0</config>
`            <guid>00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00</guid>
`        </driver>

However I don't receive any message from the nodes connected to the CAN bus.

Server log says:
`2017-12-27T22:01:19 | General | Normal | [Driver] - Preparing: VSCP_Level_II_SocketCAN_Driver_A
`2017-12-27T22:01:19 | General | Normal | [Driver] - Starting drivers...
`2017-12-27T22:01:21 | General | Normal | [Device tread] Unable to open VSCP driver (check username/password/path/rights).

The nodes and the can bus works fine with VSCP server 1.0.1.7 Neon.

Thank you

grodansparadis, 2017-12-28:
> |<config>can0</config>| 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00

Hi,

just tested and things works here.  Sone wuestions to be able to 
invesigate firther. If you are running a non daemon session run what is 
the output?  Drivers log additional log info to the syslog file do you 
have some info there?

/Ake

mamac, 2017-12-28:
Below output when running vscpd in interactive mode:

> Prepare to start vscpd...
Stay Hungry. Stay Foolish.
I will ***NOT*** run as daemon! (ctrl+c to terminate)
[vscpd] Configfile = /etc/vscp/vscpd.conf
VSCPD: init.
Reading XML GENERAL configuration from [/etc/vscp/vscpd.conf]
Using configuration file: /etc/vscp/vscpd.conf
2017-12-28T15:41:33 - Reading full XML configuration from [/etc/vscp/vscpd.conf]
2017-12-28T15:41:34 - Level II driver added. - [ /srv/vscp/drivers/level2/vscpl2drv-socketcan.so ]
2017-12-28T15:41:34 - Level II driver added. - [ /srv/vscp/drivers/level2/vscpl2drv-sim.so ]
2017-12-28T15:41:34 - Reading VSCP table xml info: Invalid table type (static/dynamic)!
2017-12-28T15:41:34 - Reading VSCP table xml info: Invalid table type (static/dynamic)!
2017-12-28T15:41:34 - loading users from users db...
2017-12-28T15:41:34 - VSCP Server started
Version: 12.29.2.30 Magnesium (Development version)
Copyright (C) 2000-2017, Grodans Paradis AB, http://www.paradiseofthefrog.com
2017-12-28T15:41:34 - Log Level=1
2017-12-28T15:41:34 - Reading in user tables from DB.
2017-12-28T15:41:34 - Load tables: Loading tables from database :
2017-12-28T15:41:34 - Initializing user tables.
2017-12-28T15:41:34 - Initializing DM.
2017-12-28T15:41:34 - Loading DM.
2017-12-28T15:41:35 - Initialize variables.
2017-12-28T15:41:35 - Loading persistent variables from XML variable default path.
2017-12-28T15:41:36 - Starting client worker thread...
2017-12-28T15:41:36 - Starting TCP/IP interface...
2017-12-28T15:41:36 - Starting UDP simple interface...
2017-12-28T15:41:36 - Starting multicast channel interface thread...
2017-12-28T15:41:36 - UDP Client: Thread started.
2017-12-28T15:41:36 - [Multicast channel] Unable to add to multicast group.
2017-12-28T15:41:36 - [Driver] - Starting drivers...
2017-12-28T15:41:36 - [Driver] - Preparing: VSCP_Level_II_SocketCAN_Driver_A
2017-12-28T15:41:36 - [Driver] - Started driver .VSCP_Level_II_SocketCAN_Driver_A
2017-12-28T15:41:36 - [Driver] - Preparing: VSCP_Level_II_sim_driver
2017-12-28T15:41:36 - Loading level II driver: <VSCP_Level_II_SocketCAN_Driver_A>
2017-12-28T15:41:36 - [Driver] - Started driver .VSCP_Level_II_sim_driver
2017-12-28T15:41:36 - Discovered all methods
2017-12-28T15:41:36 - Starting VSCP Server worker thread,,.
VSCPD: run.
2017-12-28T15:41:36 - Loading level II driver: <VSCP_Level_II_sim_driver>
2017-12-28T15:41:36 - [TCP/IP srv] -- Accept.
2017-12-28T15:41:36 - Discovered all methods
2017-12-28T15:41:36 - [TCP/IP srv] -- Accept.
2017-12-28T15:41:38 - [TCP/IP srv] Host [127.0.0.1] User [drv_UXFWVVMQmTH999575MByqd0ZRXjhqB7K] allowed to connect.
2017-12-28T15:41:38 - [TCP/IP srv] Command: Close.
2017-12-28T15:41:38 - [Device tread] Unable to open VSCP driver (check username/password/path/rights).
2017-12-28T15:41:39 - [Device tread] Unable to open VSCP driver (check username/password/path/rights).
2017-12-28T15:41:39 - [TCP/IP srv] Command: Close.`

Why do we have "Unable to open VSCP driver "?

mamac, 2017-12-28:
And the output from syslog is pretty much the same:
>vscp@lljm2:~ $ cat /var/log/syslog |grep -i vscp|grep -i socket
Dec 28 13:36:51 lljm2 vscpd: 2017-12-28T13:36:51 - Level II driver added. - [ /srv/vscp/drivers/level2/vscpl2drv-socketcan.so ]
Dec 28 13:36:54 lljm2 vscpd: 2017-12-28T13:36:54 - [Driver] - Preparing: VSCP_Level_II_SocketCAN_Driver_A
Dec 28 13:36:54 lljm2 vscpd: 2017-12-28T13:36:54 - [Driver] - Started driver .VSCP_Level_II_SocketCAN_Driver_A
Dec 28 13:36:54 lljm2 vscpd: 2017-12-28T13:36:54 - VSCP_Level_II_SocketCAN_Driver_A Started at 2017-12-28 13:36:54
Dec 28 13:36:54 lljm2 vscpd: 2017-12-28T13:36:54 - Loading level II driver: <VSCP_Level_II_SocketCAN_Driver_A>
Dec 28 13:43:37 lljm2 vscpd: 2017-12-28T13:43:37 - Level II driver added. - [ /srv/vscp/drivers/level2/vscpl2drv-socketcan.so ]
Dec 28 13:43:39 lljm2 vscpd: 2017-12-28T13:43:39 - [Driver] - Preparing: VSCP_Level_II_SocketCAN_Driver_A
Dec 28 13:43:39 lljm2 vscpd: 2017-12-28T13:43:39 - [Driver] - Started driver .VSCP_Level_II_SocketCAN_Driver_A
Dec 28 13:43:39 lljm2 vscpd: 2017-12-28T13:43:39 - VSCP_Level_II_SocketCAN_Driver_A Started at 2017-12-28 13:43:39
Dec 28 13:43:39 lljm2 vscpd: 2017-12-28T13:43:39 - Loading level II driver: <VSCP_Level_II_SocketCAN_Driver_A>
Dec 28 13:47:06 lljm2 vscpd: 2017-12-28T13:47:06 - Level II driver added. - [ /srv/vscp/drivers/level2/vscpl2drv-socketcan.so ]
Dec 28 13:47:09 lljm2 vscpd: 2017-12-28T13:47:09 - [Driver] - Preparing: VSCP_Level_II_SocketCAN_Driver_A
Dec 28 13:47:09 lljm2 vscpd: 2017-12-28T13:47:09 - [Driver] - Started driver .VSCP_Level_II_SocketCAN_Driver_A
Dec 28 13:47:09 lljm2 vscpd: 2017-12-28T13:47:09 - VSCP_Level_II_SocketCAN_Driver_A Started at 2017-12-28 13:47:09
Dec 28 13:47:09 lljm2 vscpd: 2017-12-28T13:47:09 - Loading level II driver: <VSCP_Level_II_SocketCAN_Driver_A>
Dec 28 13:51:39 lljm2 vscpd: 2017-12-28T13:51:39 - Level II driver added. - [ /srv/vscp/drivers/level2/vscpl2drv-socketcan.so ]
Dec 28 13:51:43 lljm2 vscpd: 2017-12-28T13:51:43 - [Driver] - Preparing: VSCP_Level_II_SocketCAN_Driver_A
Dec 28 13:51:43 lljm2 vscpd: 2017-12-28T13:51:43 - [Driver] - Started driver .VSCP_Level_II_SocketCAN_Driver_A
Dec 28 13:51:43 lljm2 vscpd: 2017-12-28T13:51:43 - VSCP_Level_II_SocketCAN_Driver_A Started at 2017-12-28 13:51:43
Dec 28 13:51:43 lljm2 vscpd: 2017-12-28T13:51:43 - Loading level II driver: <VSCP_Level_II_SocketCAN_Driver_A>
Dec 28 14:26:07 lljm2 vscpd: 2017-12-28T14:26:07 - Level II driver added. - [ /srv/vscp/drivers/level2/vscpl2drv-socketcan.so ]
Dec 28 14:26:11 lljm2 vscpd: 2017-12-28T14:26:11 - [Driver] - Preparing: VSCP_Level_II_SocketCAN_Driver_A
Dec 28 14:26:11 lljm2 vscpd: 2017-12-28T14:26:11 - [Driver] - Started driver .VSCP_Level_II_SocketCAN_Driver_A
Dec 28 14:26:11 lljm2 vscpd: 2017-12-28T14:26:11 - VSCP_Level_II_SocketCAN_Driver_A Started at 2017-12-28 14:26:11
Dec 28 14:26:11 lljm2 vscpd: 2017-12-28T14:26:11 - Loading level II driver: <VSCP_Level_II_SocketCAN_Driver_A>
Dec 28 15:41:34 lljm2 vscpd: 2017-12-28T15:41:34 - Level II driver added. - [ /srv/vscp/drivers/level2/vscpl2drv-socketcan.so ]
Dec 28 15:41:36 lljm2 vscpd: 2017-12-28T15:41:36 - [Driver] - Preparing: VSCP_Level_II_SocketCAN_Driver_A
Dec 28 15:41:36 lljm2 vscpd: 2017-12-28T15:41:36 - [Driver] - Started driver .VSCP_Level_II_SocketCAN_Driver_A
Dec 28 15:41:36 lljm2 vscpd: 2017-12-28T15:41:36 - VSCP_Level_II_SocketCAN_Driver_A Started at 2017-12-28 15:41:36
Dec 28 15:41:36 lljm2 vscpd: 2017-12-28T15:41:36 - Loading level II driver: <VSCP_Level_II_SocketCAN_Driver_A>

grodansparadis, 2017-12-28:
>Why do we have "Unable to open VSCP driver "?
This comes from that the driver can't log in into the tcp/ip interface.  A valid user/password is created when the server is started for the drivers to use and if it is wrong then the login fail. Can it be that you also have a daemon running in the background?   "pgrep vscpd"

mamac, 2017-12-28:
"pgrep vscpd" doesn't show running deamon, I killed it before running vscpd interactive.
Anyway, does the tcp/ip login issue interact with loading the socketCAN driver?

Just to make sure, I'm trying to get vscpd to use can0 interface (8devices usb/can interface)

mamac, 2017-12-28:
Re-reading vscpd.conf I found this strange lines:
>        <driver enable="false" >
            <name>VSCP Level II tcpiplink Driver  A</name>
            <path>/srv/vscp/drivers/level2/vscpl2drv-tcpiplink.so</path>
            <config>192.168.1.2;9598;admin;secret</config>
            <guid>00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00</guid>
        </driver>

is it normal to have password in clear text?

grodansparadis, 2017-12-28:
For a tcp/ip driver that should log in on a remote server you will have to use this format. No it is not perfect and you should of course definitely not normally use the admin credentials, but you can limit just this driver access from only this machine with the set credentials so it is normally not such a bad thing.

grodansparadis, 2017-12-28:
Level I drivers work directly against the interface (you can try the Level I socketcan version). Level II drivers log in to the tcp/ip interface of the daemon with the user/password pair I described earlier. This also means that local access to tcp/ip must be allowed. 

As I said I saw this before Christmas also but never investigated it. But it can have been due to a daemon running in the background at the time. But also due to some problem fix (i'm at build 31 and you at build 30). Worst case is that it should be an intermittent error.

mamac, 2017-12-28:
ok I'll try with the level I driver

grodansparadis, 2017-12-28:
btw Do you get events from the sim driver?

mamac, 2017-12-28:
What does sim driver event look like?

Trying to connect to vscpd remotely with WORKS from my computer and it fails:

`2017-12-28T17:21:44 - [TCP/IP srv] -- Accept.`
`2017-12-28T17:21:46 - [TCP/IP srv] Host [192.168.1.123] User [admin] allowed to connect.`
`2017-12-28T17:21:47 - [TCP/IP srv] Command: Close.`

Even though it was working yesterday night... playing around too much with vscpd.conf...

mamac, 2017-12-28:
In the config it says:
>     

How do you use mkpassword?

Maybe I need to set user's password properly.

grodansparadis, 2017-12-28:
It will send a temperature event every second.  It fetch configuration (as most Level II drivers) from the tcp/ip interface. The default variable xml file contains them.  Here you have my conf for Level II drivers

<level2driver enable="true">

        
        <driver enable="false" >
            <name>VSCP Level II Logger Driver A</name>
            <path>/srv/vscp/drivers/level2/vscpl2drv-logger.so</path>
             <config>/tmp/vscp_level2.log</config>
             <guid>00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00</guid>
        </driver>

        
        <driver enable="false" >
            <name>VCCP Level II LM-sensors driver A</name>
            <path>/srv/vscp/drivers/level2/vscpl2drv-lmsensors.so</path>
            <config>2</config>
            <guid>00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00</guid>
        </driver>

        
        <driver enable="false" >
            <name>VSCP Level II SocketCAN Driver A</name>
            <path>/srv/vscp/drivers/level2/vscpl2drv-socketcan.so</path>
            <config>vcan0</config>
            <guid>00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00</guid>
        </driver>

        
        <driver enable="false" >
            <name>VSCP Level II MQTT Driver A</name>
            <path>/srv/vscp/drivers/level2/vscpl2drv-mqtt.so</path>
            <config>mysession;subscribe;vscp;localhost;1883;;60</config>
            <guid>08:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00</guid>
        </driver>

        
        <driver enable="false" >
            <name>VSCP Level II tcpiplink Driver  A</name>
            <path>/srv/vscp/drivers/level2/vscpl2drv-tcpiplink.so</path>
            <config>192.168.1.2;9598;admin;secret</config>
            <guid>00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00</guid>
        </driver>

        
        <driver enable="false">
            <name>VSCP Level II RawEthernet Driver A</name>
            <config>eth0</config>
            <path>/srv/vscp/drivers/level2/vscpl2drv-raweth.so</path>
        </driver>

       
        <driver enable="true">
            <name>VSCP Level II sim driver</name>  
            <config>1</config>
            <guid>FF:FF:FF:FF:FF:FF:FF:F7:03:00:00:00:00:00:00:00</guid>
            <path>/srv/vscp/drivers/level2/vscpl2drv-sim.so</path>
        </driver>

        
        <driver enable="false">
            <name>VSCP Level II 1-wire Driver A</name>
            <config>eth0</config>
            <path>/srv/vscp/drivers/level2/vscpl2drv-wire1.so</path>
        </driver>

    </level2driver>

grodansparadis, 2017-12-28:
mkpasswd 'password' 

will generate the safe key and hash for you.   You can do this in the web interface also now.

grodansparadis, 2017-12-28:
And its vscp-mkpasswd now

mamac, 2017-12-28:
I reset password and I can connect remotely to the deamon with WORKS. :-)
Let's see if I got events from the nodes connected to the CAN bus.

mamac, 2017-12-28:
When level II socketCAN driver is enabled I dont get any event except events generated by the deamon.

No event from de sim driver.

If I enable the level I driver for socketCAN, then I cannot login to vscpd using WORKS from my laptop so I can't see if events are received, and I don't see any event received looking at the interactive mode.

grodansparadis, 2017-12-28:
In the last case can it be that the XML file gets corrupted. The server will not start if it is.  Or does it run when you enable the level I driver?

mamac, 2017-12-28:
The server runs with the level I enabled.

Could it be that the latest version of VSCPD is not compatible with messages sent by the nodes?

grodansparadis, 2017-12-28:
No the events are the same it's only in the tcp/ip interface the time information has been added.

I can't understand why you can't login when you enable the Level I driver. The two are not related in  any way. Strange!

mamac, 2017-12-28:
I don't get it either.
would you my to send me your vscpd.conf?
As you said maybe it's corrupted...

mamac, 2017-12-28:
Another test I did:
- connect to current vscp server with WORKS
- send TURNON event to the bus manually
-> node reacts

- connect to new vscp server with WORKS
- send TURNON event to the bus manually
-> no reaction from the node

Therefore there is clearly no communication between the deamon and the CAN bus.

mamac, 2017-12-28:
I installed VSCP on another raspberry and there everything works fine.
There must be something wrong on the first raspberry, I'll redo the setup.
Sorry for spending your time...

closed 2017-12-28 by mamac

grodansparadis, 2017-12-28:
No problem. If you experience eventually someone else will.  The symptoms was very strange indeed.

grodansparadis, 2017-12-28:
Hopefully Level II drivers will be able possible to add/start and stop from the web interface in the version after this one. I also add some more possibilities for debugging as we go.