grodansparadis/vscp #235: Web socket through SSL

mamac, 2018-01-01:
Hi,

Trying to get websocket to work through secured connection.

Setup is as below:

` var vscp_setting_url="wss://192.168.1.2:443";		// SSL`

The output from the browser is this:

>Firefox can’t establish a connection to the server at wss://192.168.1.3/.  vscpwslib.js:1590:13
>Close VSCP websocket  vscpwslib.js:5091:23
>time4VariableRead  vscpwslib.js:3422:23
>InvalidStateError: An attempt was made to use an object that is not, or is no longer, usable  vscpwslib.js:3424
>Firefox can’t establish a connection to the server at wss://192.168.1.3/

BlueAndi, 2018-01-01:
The current source head uses port 8843, instead of 443.

mamac, 2018-01-01:
Thanks a lot!

closed 2018-01-01 by mamac