grodansparadis/vscp #236: DM not working

mamac, 2018-01-01:
Hello,

Sorry to raise an issue again.

On a fresh implementation of VSCP on a raspberry PI, I cannot make the decision matrix to work.

Nothing written in syslog regarding DM.

vscp.conf:
>    <dm enable="true"
          path="/srv/vscp/dm.xml"

I can see the events received by the daemon, but nothing happens, even when running "vscpd -s" for interactive mode.

>Jan  1 23:25:34 lljm vscpd: 2018-01-01T23:25:34 - Reading full XML configuration from [/etc/vscp/vscpd.conf]
>Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Level II driver added. - [ /srv/vscp/drivers/level2/vscpl2drv-socketcan.so ]
>Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Reading VSCP table xml info: Invalid table type (static/dynamic)!
>Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Reading VSCP table xml info: Invalid table type (static/dynamic)!
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - loading users from users db...
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Ethernet MAC address: B8:27:EB:F8:B8
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - VSCP Server started#012Version: 12.29.2.30 Magnesium (Development version)#012Copyright (C) 2000-2017, Grodans Paradis AB, http://www.paradiseofthefrog.com
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Log Level=1
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Reading in user tables from DB.
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Load tables: Loading tables from database :
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Initializing user tables.
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Initializing DM.
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Loading DM.
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - [DM] Loading decision matrix from: /srv/vscp/dm.xml
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Initialize variables.
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - [Variable database init] Path=/srv/vscp/variable.sqlite3
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Loading persistent variables from XML variable default path.
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Starting client worker thread...
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Starting TCP/IP interface...
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Starting UDP simple interface...
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - [TCP/IP srv] Thread started.
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Starting multicast channel interface thread...
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - UDP Client: Bind to interface. [udp://44444]
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - [Multicast channel] Unable to add to multicast group.
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - UDP Client: Thread started.
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - [Driver] - Starting drivers...
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - [Driver] - Preparing: VSCP_Level_II_SocketCAN_Driver_A
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - [Driver] - Started driver .VSCP_Level_II_SocketCAN_Driver_A
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - VSCP_Level_II_SocketCAN_Driver_A Started at 2018-01-01 23:25:35
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Starting VSCP Server worker thread,,.
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Loading level II driver: <VSCP_Level_II_SocketCAN_Driver_A>
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Discovered all methods
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - [TCP/IP srv] -- Accept.
Jan  1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - [TCP/IP srv] Ready to serve client.
Jan  1 23:25:36 lljm vscpd: 2018-01-01T23:25:36 - [TCP/IP srv] Host [127.0.0.1] User [drv_kzsVwCtybMMxee3KEkMR9s4bgVfdsEYd] allowed to connect.
Jan  1 23:25:36 lljm vscpd: 2018-01-01T23:25:36 - [TCP/IP srv] Command: Close.
Jan  1 23:25:44 lljm vscpd: 2018-01-01T23:25:44 - [websrv] 192.168.1.123 - - [01/Jan/2018:23:25:44 +0100] "GET /vscp/ HTTP/1.1" 1 6740 http://192.168.1.2:8884/vscp/dm Mozilla/5.0 (X11; Linux x86_64; rv:52.0) Gecko/20100101 Firefox/52.0
Jan  1 23:25:52 lljm vscpd: 2018-01-01T23:25:52 - [websrv] 192.168.1.123 - - [01/Jan/2018:23:25:51 +0100] "GET /vscp/dm HTTP/1.1" 1 13477 http://192.168.1.2:8884/vscp/ Mozilla/5.0 (X11; Linux x86_64; rv:52.0) Gecko/20100101 Firefox/52.0
Jan  1 23:26:33 lljm vscpd: 2018-01-01T23:26:33 - [TCP/IP srv] -- Accept.
Jan  1 23:26:33 lljm vscpd: 2018-01-01T23:26:33 - [TCP/IP srv] Ready to serve client.
Jan  1 23:26:33 lljm vscpd: 2018-01-01T23:26:33 - [TCP/IP srv] -- Accept.
Jan  1 23:26:33 lljm vscpd: 2018-01-01T23:26:33 - [TCP/IP srv] Ready to serve client.
Jan  1 23:26:34 lljm vscpd: 2018-01-01T23:26:34 - [TCP/IP srv] Host [192.168.1.123] User [admin] allowed to connect.
Jan  1 23:26:34 lljm vscpd: 2018-01-01T23:26:34 - [TCP/IP srv] Host [192.168.1.123] User [admin] allowed to connect.
Jan  1 23:27:15 lljm vscpd: 2018-01-01T23:27:15 - [websrv] 192.168.1.123 - - [01/Jan/2018:23:27:14 +0100] "GET /vscp/dmedit?new=true HTTP/1.1" 1 11895 http://192.168.1.2:8884/vscp/dm Mozilla/5.0 (X11; Linux x86_64; rv:52.0) Gecko/20100101 Firefox/52.0
Jan  1 23:29:42 lljm vscpd: 2018-01-01T23:29:42 - [websrv] 192.168.1.123 - - [01/Jan/2018:23:29:41 +0100] "GET /vscp/dmpost?groupid=&id=-1&new=true&from=0&count=50&filter_priority=-1&mask_priority=0x00&filter_vscpclass=+20&mask_vscpclass=0xFFFF&filter_vscptype=9+&mask_vscptype=0xFFFF&filter_vscpguid=00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00&mask_vscpguid=00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00%3A00&vscpindex=+&vscpzone=0&vscpsubzone=0&check_enablerow=true&check_index=true&check_zone=true&check_subzone=true&allowedfrom=*&allowedto=*&allowedtime=*&monday=true&tuesday=true&wednesday=true&thursday=true&friday=true&saturday=true&sunday=true&action=0x10&actionparameter=%2Fusr%2Fbin%2Ftouch+%2Fsrv%2Fvscp%2Ftestdm.txt&comment=&btnsave=btnsave HTTP/1.1" 1 6512 http://192.168.1.2:8884/vscp/dmedit?new=true Mozilla/5.0 (X11; Linux x86_64; rv:52.0) Gecko/20100101 Firefox/52.0
Jan  1 23:29:45 lljm vscpd: 2018-01-01T23:29:45 - [websrv] 192.168.1.123 - - [01/Jan/2018:23:29:44 +0100] "GET /vscp/dm?from=0&count=50 HTTP/1.1" 1 14683 - Mozilla/5.0 (X11; Linux x86_64; rv:52.0) Gecko/20100101 Firefox/52.0
Jan  1 23:31:11 lljm vscpd: 2018-01-01T23:31:11 - [DM] Saving decision matrix to: /srv/vscp/dm.xml
Jan  1 23:31:11 lljm vscpd: 2018-01-01T23:31:11 - [websrv] 192.168.1.123 - - [01/Jan/2018:23:31:11 +0100] "GET /vscp/dmdelete?id=4 HTTP/1.1" 1 6501 http://192.168.1.2:8884/vscp/dm?from=0&count=50 Mozilla/5.0 (X11; Linux x86_64; rv:52.0) Gecko/20100101 Firefox/52.0
Jan  1 23:31:14 lljm vscpd: 2018-01-01T23:31:14 - [websrv] 192.168.1.123 - - [01/Jan/2018:23:31:11 +0100] "GET /vscp/dm?from=0&count=50 HTTP/1.1" 1 13477 - Mozilla/5.0 (X11; Linux x86_64; rv:52.0) Gecko/20100101 Firefox/52.0
Jan  1 23:31:30 lljm vscpd: VSCPD: signal received, forced to stop.: Success
Jan  1 23:31:30 lljm vscpd: 2018-01-01T23:31:30 - VSCPD: signal received, forced to stop.
Jan  1 23:31:37 lljm vscpd: 2018-01-01T23:31:37 - [TCP/IP srv] Free.
Jan  1 23:31:37 lljm vscpd: 2018-01-01T23:31:37 - [TCP/IP srv] Quit.
Jan  1 23:31:37 lljm vscpd: 2018-01-01T23:31:37 - [TCP/IP srv] Exit.
Jan  1 23:31:37 lljm vscpd: 2018-01-01T23:31:37 - UDP ClientThread: Quit.
Jan  1 23:31:37 lljm vscpd: VSCPD: Will end things.

Anywhere I can trace an error message?

grodansparadis, 2018-01-02:
Looks OK as far as I can see.   

You have DM logging in the system log. You can investigate it in the admin web interface (http://localhost:8884/vscp/log), select Decision Matrix log. As default there is a minimum of debug info. in the log. You can begin with setting [loglevel](http://www.vscp.org/docs/vscpd/doku.php?id=configuring_the_vscp_daemon#loglevel) to 2 for maximum log info. If you have a late version you can also set [logflags1](http://www.vscp.org/docs/vscpd/doku.php?id=configuring_the_vscp_daemon#logflags1) to 32 so you get really great amount of log info from the DM handling.

In the administration web interface (http://localhost:8884/vscp/dm) you can see when a row is triggered (Trigger count) and if an error is generated (error count). 

If this info does not help please post the XML (or screen dump) for the DM row you have problem with.

mamac, 2018-01-02:
Thank you Ake for fast answer.

The problem is that it occures on all DM rows and there is nothing in the log related to DM.
Also trigger count and error count is set to zero on esch DM rows:
[image]

It's like DM feature is not activated even if it is:
>Jan 1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Initializing DM.
Jan 1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - Loading DM.
Jan 1 23:25:35 lljm vscpd: 2018-01-01T23:25:35 - [DM] Loading decision matrix from: /srv/vscp/dm.xml

Therefore I'm searching a way to get extra log info in order to know what happens.

I attach dm.xml in txt format since xml is not supported in github issues.
 
[dm.xml.txt](https://github.com/grodansparadis/vscp/files/1598919/dm.xml.txt)

grodansparadis, 2018-01-03:
The DM looks OK as far as I can see. The only possible problem may be the lines 

```
    <index  bMeasurement="false"  > 0</index>
    <zone>0</zone>
    <subzone>0</subzone>
```

which force a check for the items. You can safely remove them but maybe add  the index line later with **bMeasurement** set to true and the sensor index set to the sensor index of interest.  But actually I don't think this is the problem as they are all at default values above.

Are you sure that you receive CLASS=10, Type=6 events (first DM entry) ?   If booth the trigger and the error counter is zero it appears as you don't. 

To see that rows actually trigger try one of the [VSCP server internal events](http://www.vscp.org/docs/vscpspec/doku.php?id=class2.vscpd) For example the [minute](http://www.vscp.org/docs/vscpspec/doku.php?id=class2.vscpd#type_6_0x0006_minute) or second event

mamac, 2018-01-03:
Below is a screenshot of what the deamon receives:
[image]
So it confirms that events are actually received by the deamon.

I will play a bit with internal events to see how it goes.

grodansparadis, 2018-01-03:
> image

Did you remove the index etc rows?

grodansparadis, 2018-01-03:
btw edit the DM in the XML file. I found some errors in the web editing 
today and is fixing this now.

mamac, 2018-01-03:
I deleted the whole dm.xml file and restarted the server to start with a fresh file, but apparently it doesn't generate a new file.
Would you have a working dm.xml file?

mamac, 2018-01-03:
Redid a test with a simple dm.xml file:

<?xml version = "1.0" encoding = "UTF-8" ?>
<dm>
  <row enable="true" groupid="" >
    <mask  priority="0"  class="65535"  type="65535"  GUID=" 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:FF" > </mask>
    <filter  priority="0"  class="65535"  type="5"  GUID=" 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:01" > </filter>
    <action>0x80</action>
    <param>SOUTHTEMP;%unixtime;%measurement.float</param>
    <comment>Store measurements in table</comment>
    <allowed_from>0000-01-01 00:00:00</allowed_from>
    <allowed_to>9999-12-31 23:59:59</allowed_to>
    <allowed_weekdays>mtwtfss</allowed_weekdays>
    <allowed_time>*-*-* *:*:*</allowed_time>
    <index  bCheckMeasurementIndex="false"  > 1</index>
    <zone>0</zone>
    <subzone>0</subzone>
    <measurement enable="false" compare="noop" unit="0" value="0.000000"  />
  </row>

</dm>

But still nothing in the log...

grodansparadis, 2018-01-03:
You have the distribution file in the config_examples folder of the source.

Attached is the file I use [for this example](https://thingspeak.com/channels/79008).  Basically it uploads some values to Thingspeak every five minutes.

[dm.xml.txt](https://github.com/grodansparadis/vscp/files/1601831/dm.xml.txt)

grodansparadis, 2018-01-03:
The row   

<index bCheckMeasurementIndex="false">1</index>

will not probably do what you expect. On the form it is written no it will trigger only if byte 0 of the data (index) is set to one (not there for Class=0xffff as it have no data).  If you leave the line out altogether this check will not be done. But I suspect that you want to check for sensorindex=1 and that you will do by using the form you have now with bCheckMeasurementIndex set to true.  But for class=65535 it should also be left out, But valid for a measurement (class=10).

The row <zone>0</zone> require  that byte 1 of the data is 0 so remove.

The row <subzone>0</subzone> require  that byte 2 of the data is 0 so remove.

The measurement row is not needed.

I recommend using action= 0 (noop)  until you see that a row triggers.

mamac, 2018-01-03:
I took your d.xml example and changed is to have only one row:
[image]

It will do something when receiving class 20 type 9 events.

I receive 20/9 events but still nothing in the DM log...

I saw there are new changes on the master branch, I'll download it and compile it to see if it gets any better.

grodansparadis, 2018-01-03:
Well that row is **disabled** so it will not do anything.

mamac, 2018-01-03:
Indeed... so I enabled it, but still nothing happens:
[image]

Building the latest source from the master branch.

grodansparadis, 2018-01-04:
As long as you edited the "enable" in the xml file and not in the admin web i/f it should work and I don't have a clue why it does not work for you.

The reason you can't save DM rows in the admin web i/f is that I discovered yesterday that three fields that was added in the new version is not handled in the admin web i/f. This have the effect that <index>, <zone>, <subzone> is written even if they are not used. In turn meaning that checks will be done for these values to trigger a row. I am working on this now.

Also instead of syslog use the DM log to see what is happening in the DM. Go to

`http://your-host:8884/vscp/log`

and select the DM alternative. But you will not find a lot of info there if the row does not get triggered and also not if you don't set the debug flags for the DM.

mamac, 2018-01-04:
Built the latest source code from the master branch, reloaded all default configuration files, enabled the debug log level, but still nothing:
[image]

grodansparadis, 2018-01-04:
Can you send the XML file also please?

grodansparadis, 2018-01-04:
The "File does not exist" in the log looks strange. Is the dm.xml file where you think it is? Is it readable by the vscp server?  The four equal entries in a row is also strange? Only one instance started?

mamac, 2018-01-04:
I deleted the dm.xml file yesterday to see what happens, sometimes it helps to generate more errors...
The four equal entries are because I restarted vscpd everytime I updated dm.xml

grodansparadis, 2018-01-04:
OK

mamac, 2018-01-04:
I did another test. I ask the DM to run an external command:
[image]
Still no luck.
Sorry to spend your time for this...

mamac, 2018-01-04:
Attaching the dm.xml I am working with.
[dm.xml.txt](https://github.com/grodansparadis/vscp/files/1603279/dm.xml.txt)

grodansparadis, 2018-01-04:
No problem with spending my time. This is my "job". ;-)  Anyway I am glad to try to help.

This is my results from your file

[image]

I changed the action to noop and trigger on CLASS2.VSCPD, Type = 5   I also changed the GUID mask to all zeros so no check is done there .

You can test it

[dm.xml.txt](https://github.com/grodansparadis/vscp/files/1603399/dm.xml.txt)

mamac, 2018-01-04:
I took your xml and now I got trigger count.
But still the file is not created and nothing in the log:
[image]

vscpd runs with user "vscp", and user "vscp" is allowed to create a file in /srv/vscp ( I tested maunally)

grodansparadis, 2018-01-04:
Try to change the action to 16

mamac, 2018-01-04:
Off course.
Now it works, the file is created.
However the log file doesn't show that the action took place, is that normal?

I remember on my previous server I could see all actions triggered by the DM in the log file.

mamac, 2018-01-04:
Could it be that the "vscp" user running the deamon is not allowed to write in the log file?

grodansparadis, 2018-01-04:
Did you set the debug flags as described above?

mamac, 2018-01-04:
I set the debug flag in vscpd.conf, is there any other place to set it?

grodansparadis, 2018-01-04:
**logflags1** should be **debugflags1**  sorry

grodansparadis, 2018-01-04:
Also you have to set debug/2 as loglevel    

The default log information is drastically lowered in this version to save disk.

mamac, 2018-01-04:
I'll see if I can get more information from the log.
At least the DM now works. Thank you!

closed 2018-01-04 by mamac

grodansparadis, 2018-01-04:
Good!  Eventually (if eons of time lays ahead) I hope there will be user friendly wizards for this. Making things easy. The XML is still not intended  for humans.