grodansparadis/vscp #237: ssleay32.dll missing in vscpworks-13.0.0.4-x64.exe

BlueAndi, 2018-03-13:
In the installation package the ssleay32.dll is missing.

grodansparadis, 2018-03-13:
Can confirm this. I will post a hotfix later tonight.

BlueAndi, 2018-03-14:
Please add the crc.h to the include folder as well.

grodansparadis, 2018-03-14:
In the VSCP Works install?

BlueAndi, 2018-03-14:
Yes. I tried to use the header files in the include folder of the VSCPWorks delivery ... anyway. Today I pushed several fixes in there.

grodansparadis, 2018-03-14:
The previous install installed everything. Will go for separate 
installations of drivers, helperlib and vscp works so I guess crc.c 
should be in the helperlib install and not in vscpworks install.

Yes I noticed that you are well again. :)

BlueAndi, 2018-03-14:
Unfortunately I am ill again. :-(

I meant crc.h not crc.c ...

grodansparadis, 2018-03-14:
!!!! Same thing? Shit! :(

I also ment crc.h ;-/

BlueAndi, 2018-03-14:
No, till now not, just a bad cold. *puh*

grodansparadis, 2018-03-14:
That is good.

grodansparadis, 2018-03-14:
Have some problems with my Windows machine so this will take some more 
time than I first expected.

BlueAndi, 2018-03-14:
No problem. I thought today I can build it on my company laptop ... but didn't work either. :-)
Now I am sitting here with Linux VM and VSCPWorks is crashing all the time. Just open/close once or twice the e.g. client window. The problem seems to be around frmvscpsession.cpp .... but thats another issue. ;-)

grodansparadis, 2018-03-14:
Strange. No crash problem here.

Why were you unable to build on Windows? Was it beacuse of knownnodes? 
If so just remove them.

BlueAndi, 2018-03-14:
I tried to build according to BUILD_WIN32 with MSVC 2013, but the toolchain v141 is used, which means now it shall be build with MSVC 2017. But I can not easily update that on the company laptop, without influencing other projects on. So tried to it working with the v120 toolchain ... but gave up now.

Regarding vscpworks crash ... it crashes on windows and linux. On windows I just use the deployed executable and on linux I compiled it by my own. Both crashes are related to the same problem it looks like. I figured out that you fixed something in the past in RXWorkerThread::Entry(). gdb showed me a problem in TXWorkerThread::Entry() line 3065.

grodansparadis, 2018-03-14:
It can't be a mix between debug and release dll's. That will crash it. 
VSCP Works and the driver dll's need both be built for the same, release 
or debug.  Happen to me some time also when the install folder locations 
was changed.

grodansparadis, 2018-03-14:
> On linux I used |./configure --enable-debug| and run vscpworks in gdb
> to see the backtrace.
> On windows I am just using all the stuff of the
> vscpworks-13.0.0.4-x64.exe.
>
> Let's see how far I will come.

And you don't have an old vscp config that points to the wrong files.

BlueAndi, 2018-03-14:
I setup the VM today and used vscpworks there the first time after compilation. Strange ...

grodansparadis, 2018-03-14:
Strange at use it several time a day here.

grodansparadis, 2018-03-14:
What do you get if you issue "openssl version"

BlueAndi, 2018-03-14:
> OpenSSL 1.0.2g 1 Mar 2016

grodansparadis, 2018-03-14:
That is OK. I ask because environments change to 1.1 now and if it is 
installed one must use a switch in configure to link to it.

grodansparadis, 2018-03-14:
13.0.0.11 should fix this.

BlueAndi, 2018-03-14:
Ok, thanks.

closed 2018-03-14 by BlueAndi

BlueAndi, 2018-03-14:
Confirmed, runs out of the box.

grodansparadis, 2018-03-14:
Good