grodansparadis/vscp #238: Events from socketcan L1 can be received via websocket, but not sent.

BlueAndi, 2018-03-13:
It might be easy, but I don't find it.

I am on the head revision 8ec5dcdbf704fd7dae2ba22c940e3faa834c26ea.
Receiving events via websocket from CAN with socketcan L1 works perfect.
But if I send a event via websocket, it is never received on the CAN bus.

I also tried the socketcan L2 driver, but same behavior.

BTW. whats the difference between socketcan L1 and L2? CAN bus only supports L1, isn't it?

grodansparadis, 2018-03-13:
I run both here and as far as I can see they work. 

The Level II driver just have other configuration options.

BlueAndi, 2018-03-13:
Thanks for the info, then I will check my socketcan interface itself.

BlueAndi, 2018-03-14:
It was something easy ... the vcan0 was configured in vscpd.conf, but it is can0.
After changing it, it works.

closed 2018-03-14 by BlueAndi

grodansparadis, 2018-03-14:
Yes it should be vcan0 by default for the virtual adapter.

Good it worked out.