grodansparadis/vscp #239: Invalid GUID in vscpworks after scanning for devices

BlueAndi, 2018-03-15:
Use case:
* Start "Scan for devices..." via TCP/IP interface.
* Scan for 1 to 2, which founds 1 device.
* Double click on device jumps to device configuration window.
* On the can-bus everything is there, the request and all 32 responses.
* But a dialog appears that it failed to read the standard registers.

If I take a look to the used GUID 90:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:2E:00:01 it looks for me that this is wrong. Using the GUID FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:01:00:01 instead (got from the client window) works perfect.

Funny also, if I scan again, the GUID changes a little bit.

Can you confirm?

grodansparadis, 2018-03-15:
For some reason the tcp/ip interface is slower as it seams now on this version. I haven't bothered to investigate it as it can be solved with slacking the timing and it is scheduled to be rewritten (ssl & no mongoose to make helper lib MIT licensed). I currently use timings as

Command response timeout: 16000
Sleep after command. 200
Max number of retries: 3
Read/write timeout: 8000
Total register read/write timeout: 20000

Can you test and see if it makes things better.

The GUID you have is definitely wrong. Have you selected an interface for the session?  Is this a GUID you get after discover and going to the configure window?

BlueAndi, 2018-03-15:
The daemon seems to be slow, because the websocket connection is very slow too.

I used your timings, but same problem. I don't think it is a timing problem, because all events are received as I print them out to the console. But the receive_flags are 0, which means vscpworks thinks it never received one. So I guess they are filtered away somehow.

I selected the TCP/IP interface for the session. Yes, I get this invalid GUID after discovering.
The whole list box contains such invalid GUIDs.

BlueAndi, 2018-03-15:
[image]

grodansparadis, 2018-03-15:
OK I will check. For the device I have here it works as far as I 
remember. At least I have no problems with finding devices and reading 
standard registers after changing the timing. In most locations here I 
have the can4vscp adapter interfacing the CAN bus and it is very slow in 
the serial version and really but things to the test.

Strange with the slowness. Must dig into that.

grodansparadis, 2018-03-15:
> image

Yes the F8 in the beginning there is definitely wrong. Should be FF

BlueAndi, 2018-03-15:
I tried to figure the problem out, but no success yet.

BlueAndi, 2018-03-15:
Just noted that if I open the device configuration window directly with TCP/IP interface, the combobox contains again invalid GUIDs ... so its not related to the scanning process itself.

BlueAndi, 2018-03-15:
frmDeviceConfig::fetchIterfaceGUID() returns the invalid GUID:

[image]

BlueAndi, 2018-03-15:
What I get from the daemon:

> 14:25:34: Debug: 1,6,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:01:00:00,Remote UDP Client Listner. [udp://44444] Started at 2018-03-15 13:24:27
> 14:25:34: Debug: 2,2,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:02:00:00,socketcanl1 Started at 2018-03-15 13:24:27
> 14:25:34: Debug: 3,3,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:03:00:00,VSCP_Level_II_sim_driver Started at 2018-03-15 13:24:27
> 14:25:34: Debug: 4,1,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:04:00:00,Internal Server DM Client. Started at 2018-03-15 13:24:27
> 14:25:34: Debug: 5,1,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:05:00:00,Internal VSCP Daemon Worker Client. Started at 2018-03-15 13:24:27
> 14:25:34: Debug: 8,5,D0:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:08:00:00,Remote TCP/IP Server. [tcp://9598] Started at 2018-03-15 13:25:32
> 

The TCP/IP GUID seems to be bad.

grodansparadis, 2018-03-15:
Is it strange if you look in the web i/f (interfaces) also?

BlueAndi, 2018-03-15:
[image]

grodansparadis, 2018-03-15:
Sure you did't set a strange i/f. This is how it looks here

vvvv

rrr

BlueAndi, 2018-03-15:
The daemon shows it wrong and I don't what I could did wrong here. The vscpd.conf is created with make config-install and I only set the driver for socketcan L1 active and to can0. Thats all.

grodansparadis, 2018-03-15:
Strange, tested on the five VSCP servers here and can't see this.

grodansparadis, 2018-03-15:
Can you confirm that the guid is wrong also when you login into the 
tcp/ip interface and issue "interface list"

And it is the distro config for the daemon.

BlueAndi, 2018-03-15:
Login in via putty shows a good GUID:

[image]

BlueAndi, 2018-03-15:
For "interface list" I have to login first, but I don't know the commands for that.

grodansparadis, 2018-03-15:
user admin
pass secret

BlueAndi, 2018-03-15:
Now an additional vscpworks connection and interface list shows:

> interface list
> 1,6,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:01:00:00,Remote UDP Client Listner. [udp://44444] Started at 2018-03-15 13:24:27
> 2,2,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:02:00:00,socketcanl1 Started at 2018-03-15 13:24:27
> 3,3,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:03:00:00,VSCP_Level_II_sim_driver Started at 2018-03-15 13:24:27
> 4,1,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:04:00:00,Internal Server DM Client. Started at 2018-03-15 13:24:27
> 5,1,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:05:00:00,Internal VSCP Daemon Worker Client. Started at 2018-03-15 13:24:27
> 14,5,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:0E:00:00,Remote TCP/IP Server. [tcp://9598] Started at 2018-03-15 13:58:42
> 15,5,78:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:0F:00:00,Remote TCP/IP Server. [tcp://9598] Started at 2018-03-15 14:00:54
> +OK - Success.

BlueAndi, 2018-03-15:
The last one is the vscpworks connection, which again has an invalid GUID.

grodansparadis, 2018-03-15:
Yepp there it is.

BlueAndi, 2018-03-15:
Now on telnet session, one vscpworks Linux and one vscpworks windows sessions:
[image]

BlueAndi, 2018-03-15:
Let's pause this for a moment, as I think I am getting stupid here ... ;-)

grodansparadis, 2018-03-15:
This looks strange anyway. I will look into it.

BlueAndi, 2018-03-15:
vscpworks call SGID without an parameter, is that correct? Because it look like it destroys the GUID.

`
17:50:15: Debug: ------------------------------------------------------------
17:50:15: Debug: checkReturnValue:  Queue before. Responsetime = 16000. 
} *****5: Debug: ***** {Welcome to the VSCP daemon.
} *****5: Debug: ***** {Version: 13.0.0.11 Aluminium
} *****5: Debug: ***** {Copyright (C) 2000-2018, Grodans Paradis AB, http://www.paradiseofthefrog.com
} *****5: Debug: ***** {+OK - Success.
17:50:15: Debug: ------------------------------------------------------------
17:50:15: Debug: Welcome to the VSCP daemon.
17:50:15: Debug: Version: 13.0.0.11 Aluminium
17:50:15: Debug: Copyright (C) 2000-2018, Grodans Paradis AB, http://www.paradiseofthefrog.com
17:50:15: Debug: +OK - Success.
17:50:15: Debug: checkReturnValue: Command success!
17:50:15: Debug: doCommand: USER admin

17:50:15: Debug: ------------------------------------------------------------
17:50:15: Debug: checkReturnValue:  Queue before. Responsetime = 16000. 
17:50:15: Debug: ------------------------------------------------------------
17:50:15: Debug: +OK - User name accepted, password please
17:50:15: Debug: checkReturnValue: Command success!
17:50:15: Debug: doCommand: PASS secret

17:50:15: Debug: ------------------------------------------------------------
17:50:15: Debug: checkReturnValue:  Queue before. Responsetime = 16000. 
17:50:15: Debug: ------------------------------------------------------------
17:50:15: Debug: +OK - Success.
17:50:15: Debug: checkReturnValue: Command success!
17:50:15: Debug: doCommand: SGID

17:50:15: Debug: ------------------------------------------------------------
17:50:15: Debug: checkReturnValue:  Queue before. Responsetime = 16000. 
17:50:15: Debug: ------------------------------------------------------------
17:50:15: Debug: +OK - Success.
17:50:15: Debug: checkReturnValue: Command success!
17:50:15: Debug: doCommand: NOOP

17:50:15: Debug: ------------------------------------------------------------
17:50:15: Debug: checkReturnValue:  Queue before. Responsetime = 16000. 
17:50:15: Debug: ------------------------------------------------------------
17:50:16: Debug: +OK - Success.
17:50:16: Debug: checkReturnValue: Command success!
17:50:16: Debug: doCommand: INTERFACE LIST

17:50:16: Debug: ------------------------------------------------------------
17:50:16: Debug: checkReturnValue:  Queue before. Responsetime = 16000. 
17:50:16: Debug: ------------------------------------------------------------
17:50:16: Debug: 1,6,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:01:00:00,Remote UDP Client Listner. [udp://44444] Started at 2018-03-15 14:17:57
17:50:16: Debug: 2,2,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:02:00:00,socketcanl1 Started at 2018-03-15 14:17:57
17:50:16: Debug: 3,3,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:03:00:00,VSCP_Level_II_sim_driver Started at 2018-03-15 14:17:57
17:50:16: Debug: 4,1,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:04:00:00,Internal Server DM Client. Started at 2018-03-15 14:17:57
17:50:16: Debug: 5,1,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:05:00:00,Internal VSCP Daemon Worker Client. Started at 2018-03-15 14:17:57
17:50:16: Debug: 7,5,03:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:07:00:00,Remote TCP/IP Server. [tcp://9598] Started at 2018-03-15 14:20:34
17:50:16: Debug: 9,5,40:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:09:00:00,Remote TCP/IP Server. [tcp://9598] Started at 2018-03-15 16:50:14
17:50:16: Debug: +OK - Success.
17:50:16: Debug: checkReturnValue: Command success!
17:50:14: Debug: ev_handler: TCP/IP connect OK.
17:50:14: Debug: vscpd :ASSERT -  file: /usr/include/wx-3.0/wx/strvararg.h line:456 func: wxArgNormalizer cond: (argtype & (wxFormatStringSpecifier<T>::value)) == argtype msg: format specifier doesn't match argument type
17:50:14: Debug: ev_handler: TCP/IP receive [154, 154] Welcome to the VSCP daemon.
Version: 13.0.0.11 Aluminium
Copyright (C) 2000-2018, Grodans Paradis AB, http://www.paradiseofthefrog.com
+OK - Success.
.
17:50:14: Debug: Welcome to the VSCP daemon.
Version: 13.0.0.11 Aluminium
Copyright (C) 2000-2018, Grodans Paradis AB, http://www.paradiseofthefrog.com
+OK - Success.

 7:50:14: Debug: TCP/IP line: Welcome to the VSCP daemon.
[... 31 lines cut ...]
4,1,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:04:00:00,Internal Server DM Client. Started at 2018-03-15 14:17:57
5,1,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:05:00:00,Internal VSCP Daemon Worker Client. Started at 2018-03-15 14:17:57
7,5,03:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:07:00:00,Remote TCP/IP Server. [tcp://9598] Started at 2018-03-15 14:20:34
9,5,40:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:09:00:00,Remote TCP/IP Server. [tcp://9598] Started at 2018-03-15 16:50:14
+OK - Success.
.
17:50:16: Debug: 1,6,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:01:00:00,Remote UDP Client Listner. [udp://44444] Started at 2018-03-15 14:17:57
2,2,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:02:00:00,socketcanl1 Started at 2018-03-15 14:17:57
3,3,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:03:00:00,VSCP_Level_II_sim_driver Started at 2018-03-15 14:17:57
4,1,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:04:00:00,Internal Server DM Client. Started at 2018-03-15 14:17:57
5,1,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:05:00:00,Internal VSCP Daemon Worker Client. Started at 2018-03-15 14:17:57
7,5,03:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:07:00:00,Remote TCP/IP Server. [tcp://9598] Started at 2018-03-15 14:20:34
[... cut ...]
17:50:16: Debug: TCP/IP line: 5,1,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:05:00:00,Internal VSCP Daemon Wo ker Client. Started at 2018-03-15 14:17:57
17:50:16: Debug: TCP/IP line: 7,5,03:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:07:00:00,Remote TCP/IP Server. [ cp://9598] Started at 2018-03-15 14:20:34
17:50:16: Debug: TCP/IP line: 9,5,40:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:09:00:00,Remote TCP/IP Server. [ cp://9598] Started at 2018-03-15 16:50:14
 7:50:16: Debug: TCP/IP line: +OK - Success.
17:50:19: Debug: ev_handler: TCP/IP close.
`

BlueAndi, 2018-03-15:
VscpRemoteTcpIf::doCmdGetGUID() calls SGID, instead of GGID
🥇 
Could that be?

BlueAndi, 2018-03-15:
Fixed in ac7e9e637cc4cf16bd262eb607b55da3ad6edd3d

closed 2018-03-15 by BlueAndi