grodansparadis/vscp #240: vscpworks can't deal with UDP and TCP interface in parallel

BlueAndi, 2018-03-15:
This example shows that the UDP and TCP interface names starting with "Remote", which leads to a problem in frmScanforDevices::fetchIterfaceGUID().

> 18:10:41: Debug: 1,6,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:01:00:00,Remote UDP Client Listner. [udp://44444] Started at 2018-03-15 14:17:57
> 18:10:41: Debug: 2,2,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:02:00:00,socketcanl1 Started at 2018-03-15 14:17:57
> 18:10:41: Debug: 3,3,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:03:00:00,VSCP_Level_II_sim_driver Started at 2018-03-15 14:17:57
> 18:10:41: Debug: 4,1,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:04:00:00,Internal Server DM Client. Started at 2018-03-15 14:17:57
> 18:10:41: Debug: 5,1,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:05:00:00,Internal VSCP Daemon Worker Client. Started at 2018-03-15 14:17:57
> 18:10:41: Debug: 14,5,FF:FF:FF:FF:FF:FF:FF:FE:78:A5:04:F6:00:0E:00:00,Remote TCP/IP Server. [tcp://9598] Started at 2018-03-15 17:10:39

The frmScanforDevices::fetchIterfaceGUID() compares the interface names to find the right interface GUID. The problem is, that it cuts the interface name right before the first space appears.

`...
            for (unsigned int i = 0; i < ifarray.Count(); i++) {

                wxStringTokenizer tkz(ifarray[ i ], _(","));
                wxString strOrdinal = tkz.GetNextToken();
                wxString strType = tkz.GetNextToken();
                wxString strIfGUID = tkz.GetNextToken();
                wxString strDescription = tkz.GetNextToken();

                int pos;
                wxString strName;
                if (wxNOT_FOUND != (pos = strDescription.Find(_(" ")))) { // <------- Root cause!!!
                    strName = strDescription.Left(pos);
                    strName.Trim();
                }

                if (strName.Upper() == m_vscpif.m_strInterfaceName.Upper()) {

                    // Save interface GUID;
                    m_ifguid.getFromString(strIfGUID);

                    return true;
                }

            }
`

Is there a use case for this? If not lets throw the code part out. :-)

BlueAndi, 2018-03-15:
And after thinking a little bit ... what if there are several TCP/IP connections?

grodansparadis, 2018-03-15:
It is extremely hard for me to keep up with you at the moment as I am on 
other things.

but

This is needed as the GUID can change for an interface from time to time 
meaning that i the GUID is saved instead of the interface name would not 
be valid at a later time.

There is no meaning selecting an interface for tcp or udp or multicast 
just for drivers. At least not at the moment. Still it would be nice of 
course if the selection was constructed so it is general.

So the solution would be to add some special character ( "|" maybe) 
between the "started at..." and the name so the dependency on the space 
is not needed anymore.

/Ake

BlueAndi, 2018-03-16:
No problem Ake, don't feel pushed. The previous commit fixed that as you proposed.

Now the next problem appeared, because device configuration uses a separate connection, than scanning for devices. That leads to a different interface GUID.

I guess this part in the code should fix it, but doesn't:

frmScanforDevices::openConfiguration()
`
...
        //subframe->m_csw = m_interfaceType;
        subframe->m_csw.setInterface( m_vscpif.m_strHost,
                                        m_vscpif.m_strUser,
                                        m_vscpif.m_strPassword );
        subframe->m_ifguid = m_ifguid; // <---------------------- I guess this one was the idea to fix it.

        // Close our interface
        m_csw.doCmdClose(); // <-------------------------- Old connection closed

        // Connect to host
        subframe->enableInterface(); // <------------------- New connection

...
`

grodansparadis, 2018-03-16:
On 2018-03-16 12:43, Andreas Merkle wrote:
>
> No problem Ake, don't feel pushed. The previous commit fixed that as 
> you proposed.
>
I saw that. Great work!  No problem for me either but sometimes I can 
through a day away just by answering questions to keep others going. 
Happen quite often. Most of the time that is OK. Sometiimes harder.

> Now the next problem appeared, because device configuration uses a
> separate connection, than scanning for devices. That leads to a
> different interface GUID.
>
> I guess this part in the code should [...]

OK

btw1 - You should probably tell Iain on the list that you want one of 
his boards that he was so kind to offer to us. It would be nice if you 
had one to.

btw2 - Could you consider implementing your Level II bootloader in VSCP 
Works do you think?

Have a nice weekend.

-- 
Stay Hungry - Stay Foolish!
PGP: EF7161865D3E1CCE61E2F43464B570FEC2BE58AA
Åke Hedman
Brattbergavägen 17, 82050 LOS, SWEDEN
http://www.akehedman.se http://www.vscp.org http://www.grodansparadis.com

BlueAndi, 2018-03-16:
btw1) I am already in contact wit Iain, he offered me one.

btw2) Its a level 1 programmer, not for level 2. But I had in my mind it worked already with vscpworks? I will check it.

Have a nice weekend too!
Somehow here its not getting better. :-(

closed 2018-03-16 by BlueAndi