grodansparadis/vscp #241: Daemon seems to be very slow (v13.0.0)

BlueAndi, 2018-03-17:
I tried the bootloader wizard and programming one node needed over an hour. Very hard to test this way.
Measured on the can bus, between two extended register reads are most of the time 2s. :-(

grodansparadis, 2018-03-17:
No that is ridiculous. How is the setup you have for this so I can try 
to reproduce.

BlueAndi, 2018-03-17:
[image]

BlueAndi, 2018-03-17:
You can also use the service webinterface to see it:
* First discover your node.
* Show standard registers. There its slow too from my point of view.

grodansparadis, 2018-03-17:
> image

Just the beaglebone, no adapter.

BlueAndi, 2018-03-17:
The beaglebone has a shield with the CAN driver.

grodansparadis, 2018-03-17:
OK

grodansparadis, 2018-03-20:
fyi Not ignoring this but have a segfault problem in an other part that 
is a bit hard to find as it is intermittent. So please be patient.

grodansparadis, 2018-03-20:
Hi,

I have not really gone into this problem yet but I thought I should 
share some figures from my setup

PC ---> TCP/IP ----> RPi  ----> CAN4VSCP ---->  nodes

- A full register update is about 12 seconds

- A bootloading is about 15 minutes for a node over tcp/ip.

The CAN4VSCP adapter is very slow as it uses a serial connection so I 
guess a third or something with a USB adapter.

Much faster of course with a direct connection (CANAL).  Most timing is 
due to the small packages that each should be confirmed.  A solution for 
that would be to implement the bootloading algorithms right into the 
daemon also and just send the images.

But as I said the client should be rewritten so there does not make 
sense to try to optimize timing at this stage.

/Ake

grodansparadis, 2018-04-03:
I will have a look at this when the new tcp/ip code is in place.

grodansparadis, 2019-03-25:
Open if still a problem

closed 2019-03-25 by grodansparadis