grodansparadis/vscp #243: vscpworks not created upon install

mamac, 2018-06-06:
Hi,
I just downloaded the latest code from master branch but when I do 'make install', vscpworks is not created in /usr/local/bin.
Do I need to set it with .configure?

Thank you,
JM

grodansparadis, 2018-06-07:
You need to have wxGTK installed not only wxBase.

/Ake

mamac, 2018-06-07:
Can you please remind me which version?
Thank you

grodansparadis, 2018-06-07:
3.0 and higher should work.

mamac, 2018-06-07:
That's what I though, I have wxgtk 3 installed:

gentoo-jm /home/vscp/vscp # emerge -s wxgtk
  
[ Results for search key : wxgtk ]
Searching...

*  x11-libs/wxGTK
      Latest version available: 3.0.3-r300
      Latest version installed: 3.0.3-r300
      Size of files: 19,667 KiB
      Homepage:      https://wxwidgets.org/
      Description:   GTK+ version of wxWidgets, a cross-platform C++ GUI toolkit
      License:       wxWinLL-3 GPL-2 doc? ( wxWinFDL-3 )

[ Applications found : 1 ]

I can't see anything special in the make log, any hint?

grodansparadis, 2018-06-08:
Is it there (/usr/local/include)?

What do you get from

wx_config --libs
wx-config --cppflags
Output from vscp ./configure?

mamac, 2018-06-08:
/usr/local/include is not a directory.

gentoo-jm ~/vscp $ wx-config --list

    Default config is gtk3-unicode-3.0-gtk3
  Default config will be used for output
  Alternate matches:
    gtk2-unicode-3.0

gentoo-jm ~ # wx-config --libs
-L/usr/lib64 -pthread   -lwx_gtk3u_xrc-3.0-gtk3 -lwx_gtk3u_html-3.0-gtk3 -lwx_gtk3u_qa-3.0-gtk3 -lwx_gtk3u_adv-3.0-gtk3 -lwx_gtk3u_core-3.0-gtk3 -lwx_baseu_xml-3.0-gtk3 -lwx_baseu_net-3.0-gtk3 -lwx_baseu-3.0-gtk3 

gentoo-jm ~ # wx-config --cppflags
-I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__

vscp@gentoo-jm ~/vscp $ ./configure 
checking whether to enable debugging... no
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for gawk... gawk
checking for bison... bison -y
checking for gcc... gcc
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking how to run the C preprocessor... gcc -E
checking for flex... flex
checking lex output file root... lex.yy
checking lex library... -lfl
checking whether yytext is a pointer... yes
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln -s works... yes
checking whether make sets $(MAKE)... yes
checking for ranlib... ranlib
checking for dlsym in -ldl... yes
checking for pthread_create in -lpthread... yes
checking for MD5_Init in -lcrypto... yes
checking for sin in -lm... yes
checking for clock_gettime in -lrt... yes
checking for gzopen in -lz... yes
checking for SSL_free in -lssl... yes
checking for dirent.h that defines DIR... yes
checking for library containing opendir... none required
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/wait.h that is POSIX.1 compatible... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking OS.h usability... no
checking OS.h presence... no
checking for OS.h... no
checking arpa/inet.h usability... yes
checking arpa/inet.h presence... yes
checking for arpa/inet.h... yes
checking fcntl.h usability... yes
checking fcntl.h presence... yes
[... 170 lines cut ...]
PROJ_SUBDIRS : src/vscp/drivers/level1/can4vscp/linux src/vscp/drivers/level2/wire1/linux src/vscp/drivers/level1/logger/linux src/vscp/drivers/level1/can232/linux src/vscp/drivers/level1/xap/linux src/vscp/drivers/level1/socketcan/linux src/vscp/drivers/level2/logger/linux src/vscp/drivers/level2/lmsensors/linux src/vscp/drivers/level2/socketcan/linux src/vscp/drivers/level2/mqtt/linux src/vscp/drivers/level2/tcpdrv/linux src/vscp/drivers/level2/rawethernet/linux src/vscp/linuxvscpl1 src/vscp/helperlib/linux src/vscp/daemon/linux src/vscp/vscp-mkpasswd src/vscp/uvscpd src/vscp/vscpcmd src/vscp/drivers/level2/sim/linux

           VSCP configuration
--------------------------------------------
VSCP version          : 13.0.0.13 Aluminium (13.0.0)
System                : Linux gentoo-jm 4.12.5-gentoo #1 SMP Tue Aug 22 23:23:10 CEST 2017 x86_64 Intel(R) Core(TM) i5-6300U CPU @ 2.40GHz GenuineIntel GNU/Linux 
Architecture          : x86_64
VSCP GUI Build        : no
VSCP install dir      : /srv/vscp
no
build flavour         : release 
linking for wxWidgets : shared 

You can tune the compiler flags in vscp-config.
To build VSCP and friends, type 'make' followed by 'make install'

 use --enable-debug to build a debug version; release is default.
 use --enable-static to include a static version of wxWidgets.

 "make" -- builds VSCP & Friends.
 "make install" -- builds VSCP & Friends and install system.
 "make install-conf" -- install/restore configuration files.
 "make web-install" -- install/restore web demo content including websocket examples.
 "make deb" -- builds VSCP & Friends Debian install files.
 "make clean" -- clean project.

grodansparadis, 2018-06-10:
Looks OK. Strange.

Does all other binaries build except vscp works?

If you go init src/vscp/vscpworks and issue "make clean; make" what
output do you get?

mamac, 2018-06-11:
Only files below are created:

$ ls -l /usr/local/bin/
total 1108
-rwxr-xr-x 1 root root  42880 Jun 11 20:36 uvscpd
-rwxr-xr-x 1 root root 706360 Jun 11 20:36 vscpcmd
-rwxr-xr-x 1 root root 380688 Jun 11 20:36 vscp-mkpasswd

I'll go in src/vscp/vscpworks and do "make clean; make" , I'll send the output.

mamac, 2018-06-11:
Here is the output when I make vscpworks only:

vscp@gentoo-jm ~/vscp/src/vscp/vscpworks $ make
g++ -std=c++11  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DMG_ENABLE_THREADS -DMG_ENABLE_IPV6 -DMG_ENABLE_SSL  -DMG_ENABLE_GETADDRINFO -I. -I../common -I../../common -c vscpworks.cpp -o vscpworks.o
cc1plus: warning: /usr/local/include: not a directory
g++ -std=c++11  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DMG_ENABLE_THREADS -DMG_ENABLE_IPV6 -DMG_ENABLE_SSL  -DMG_ENABLE_GETADDRINFO -I. -I../common -I../../common -c dlgvscpmsg.cpp -o dlgvscpmsg.o
cc1plus: warning: /usr/local/include: not a directory
g++ -std=c++11  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DMG_ENABLE_THREADS -DMG_ENABLE_IPV6 -DMG_ENABLE_SSL  -DMG_ENABLE_GETADDRINFO -I. -I../common -I../../common -c frmmain.cpp -o frmmain.o
cc1plus: warning: /usr/local/include: not a directory
g++ -std=c++11  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DMG_ENABLE_THREADS -DMG_ENABLE_IPV6 -DMG_ENABLE_SSL  -DMG_ENABLE_GETADDRINFO -I. -I../common -I../../common -c frmvscpsession.cpp -o frmvscpsession.o
cc1plus: warning: /usr/local/include: not a directory
g++ -std=c++11  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DMG_ENABLE_THREADS -DMG_ENABLE_IPV6 -DMG_ENABLE_SSL  -DMG_ENABLE_GETADDRINFO -I. -I../common -I../../common -c dlgnewvscpsession.cpp -o dlgnewvscpsession.o
cc1plus: warning: /usr/local/include: not a directory
g++ -std=c++11  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DMG_ENABLE_THREADS -DMG_ENABLE_IPV6 -DMG_ENABLE_SSL  -DMG_ENABLE_GETADDRINFO -I. -I../common -I../../common -c dlgconfiguration.cpp -o dlgconfiguration.o
cc1plus: warning: /usr/local/include: not a directory
g++ -std=c++11  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DMG_ENABLE_THREADS -DMG_ENABLE_IPV6 -DMG_ENABLE_SSL  -DMG_ENABLE_GETADDRINFO -I. -I../common -I../../common -c dlgvscpinterfacesettings.cpp -o dlgvscpinterfacesettings.o
cc1plus: warning: /usr/local/include: not a directory
g++ -std=c++11  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DMG_ENABLE_THREADS -DMG_ENABLE_IPV6 -DMG_ENABLE_SSL  -DMG_ENABLE_GETADDRINFO -I. -I../common -I../../common -c dlgvscpfilter.cpp -o dlgvscpfilter.o
cc1plus: warning: /usr/local/include: not a directory
g++ -std=c++11  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DMG_ENABLE_THREADS -DMG_ENABLE_IPV6 -DMG_ENABLE_SSL  -DMG_ENABLE_GETADDRINFO -I. -I../common -I../../common -c vscpbootloader.cpp -o vscpbootloader.o
cc1plus: warning: /usr/local/include: not a directory
g++ -std=c++11  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DMG_ENABLE_THREADS -DMG_ENABLE_IPV6 -DMG_ENABLE_SSL  -DMG_ENABLE_GETADDRINFO -I. -I../common -I../../common -c dlgvscptrmit.cpp -o dlgvscptrmit.o
cc1plus: warning: /usr/local/include: not a directory
g++ -std=c++11  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DMG_ENABLE_THREADS -DMG_ENABLE_IPV6 -DMG_ENABLE_SSL  -DMG_ENABLE_GETADDRINFO -I. -I../common -I../../common -c frmdeviceconfig.cpp -o frmdeviceconfig.o
cc1plus: warning: /usr/local/include: not a directory
[... cut ...]  vscpeventhelper.o vscpremotetcpif.o mdf.o register.o canalconfobj.o vscpmulticast.o sockettcp.o crc.o crc8.o vscp_aes.o sqlite3.o mongoose.o fastpbkdf2.o vscpbase64.o vscpmd5.o devicebootloaderwizard.o dlgeventfilter.o   -L@top_wxbuilddir@lib -lssl -lz -lrt -lm -lcrypto -lpthread -ldl  -L/usr/lib64 -pthread   -lwx_gtk3u_xrc-3.0-gtk3 -lwx_gtk3u_html-3.0-gtk3 -lwx_gtk3u_qa-3.0-gtk3 -lwx_gtk3u_adv-3.0-gtk3 -lwx_gtk3u_core-3.0-gtk3 -lwx_baseu_xml-3.0-gtk3 -lwx_baseu_net-3.0-gtk3 -lwx_baseu-3.0-gtk3

grodansparadis, 2018-06-11:
Iit looks like vscpworks got still built from this output.  I think the
lsb says the  /usr/local/bin should be the default bindir  You can
change the builddir etc with flags in  the ./configure  step "cmake
-help"  for example "--bindir=DIR"

The "cc1plus: warning: /usr/local/include: not a directory" is a
warning.  So it  should work even with it. But somehow your system
indicate to the gnu autoconf that /ysr/local/include is there.

Talking against this is that you as of yourprev. mail get some binaries
in /usr/bin which also installs to the "bindir"

/Ake

grodansparadis, 2018-06-11:
Can you try "make install" also in the vscpworks folder

mamac, 2018-06-11:
vscpworks is created after 'sudo make install' ran from src/vscp/vscpworks

grodansparadis, 2018-06-11:
In /usr//bin or /usr/local/bin?

If it is possible could you do a make clean; make install in the vscp
root and post output.

mamac, 2018-06-11:
In /usr/local/bin.

Here is the output of 'sudo make install':

$ sudo make install           
hostname: invalid option -- 'I'
Usage: hostname [-v] {hostname|-F file}      set hostname (from file)
       domainname [-v] {nisdomain|-F file}   set NIS domainname (from file)
       hostname [-v] [-d|-f|-s|-a|-i|-y|-n]  display formatted name
       hostname [-v]                         display hostname

       hostname -V|--version|-h|--help       print info and exit

    dnsdomainname=hostname -d, {yp,nis,}domainname=hostname -y

    -s, --short           short host name
    -a, --alias           alias names
    -i, --ip-address      addresses for the hostname
    -f, --fqdn, --long    long host name (FQDN)
    -d, --domain          DNS domain name
    -y, --yp, --nis       NIS/YP domainname
    -F, --file            read hostname or NIS domainname from given file

   This command can read or set the hostname or the NIS domainname. You can
   also read the DNS domain or the FQDN (fully qualified domain name).
   Unless you are using bind or NIS for host lookups you can change the
   FQDN (Fully Qualified Domain Name) and the DNS domain name (which is
   part of the FQDN) in the /etc/hosts file.
# create folder structure
mkdir -p /srv/vscp
mkdir -p /srv/vscp/logs
mkdir -p /srv/vscp/actions
mkdir -p /srv/vscp/scripts
mkdir -p /srv/vscp/scripts/javascript
mkdir -p /srv/vscp/scripts/lua
mkdir -p /srv/vscp/ux
mkdir -p /srv/vscp/web
mkdir -p /srv/vscp/certs
mkdir -p /srv/vscp/drivers
mkdir -p /srv/vscp/drivers/level1
mkdir -p /srv/vscp/drivers/level2
mkdir -p /srv/vscp/upload
mkdir -p /srv/vscp/web/css
mkdir -p /srv/vscp/web/images
mkdir -p /srv/vscp/web/lib
mkdir -p /srv/vscp/web/testws
mkdir -p /srv/vscp/vscp
mkdir -p /srv/vscp/tables
mkdir -p /srv/vscp/mdf
====================================================
Building in dir  src/vscp/drivers/level1/can4vscp/linux
====================================================
make[1]: Entering directory '/home/vscp/vscp/src/vscp/drivers/level1/can4vscp/linux'
g++ -std=c++11 -D__LINUX__ -fPIC -D_REENTRANT  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=0 -pthread -D__LINUX__ -fPIC  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=0 -fno-var-tracking-assignments -I../../../../common -I../../../../../common  -c -o vscpl1drv-can4vscp.o vscpl1drv-can4vscp.cpp
cc1plus: warning: /usr/local/include: not a directory
g++ -std=c++11 -D__LINUX__ -fPIC -D_REENTRANT  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=0 -pthread -D__LINUX__ -fPIC  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=0 -fno-var-tracking-assignments -I../../../../common -I../../../../../common -c ../common/can4vscpobj.cpp -o can4vscpobj.o
cc1plus: warning: /usr/local/include: not a directory
g++ -std=c++11 -D__LINUX__ -fPIC -D_REENTRANT  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=0 -pthread -D__LINUX__ -fPIC  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=0 -fno-var-tracking-assignments -I../../../../common -I../../../../../common -c ../../../../common/guid.cpp -o guid.o
cc1plus: warning: /usr/local/include: not a directory
gcc -std=c99  -fPIC -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=0 -pthread -DCBC -I../../../../common -I../../../../../common -c ../../../../../common/dllist.c -o dllist.o
cc1: warning: /usr/local/include: not a directory
gcc -std=c99  -fPIC -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=0 -pthread -DCBC -I../../../../common -I../../../../../common -c ../../../../../common/crc8.c -o crc8.o
cc1: warning: /usr/local/include: not a directory
g++ -std=c++11 -D__LINUX__ -fPIC -D_REENTRANT  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=0 -pthread -D__LINUX__ -fPIC  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=0 -fno-var-tracking-assignments -I../../../../common -I../../../../../common -c ../../../../../common/com.cpp -o com.o
cc1plus: warning: /usr/local/include: not a directory
g++ -o vscpl1drv-can4vscp.so vscpl1drv-can4vscp.o can4vscpobj.o guid.o dllist.o crc8.o com.o -g -shared -fPIC -Wl,--no-undefined -L@top_wxbuilddir@lib  -lssl -lz -lrt -lm -lcrypto -lpthread -ldl  -L/usr/lib64 -pthread   -lwx_baseu-3.0-gtk3 -lwx_baseu_xml-3.0-gtk3 -lwx_baseu_net-3.0-gtk3 
g++ -std=c++11 -D__LINUX__ -fPIC -D_REENTRANT  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=0 -pthread -D__LINUX__ -fPIC  -I/usr/lib64/wx/include/gtk3-unicode-3.0-gtk3 -I/usr/include/wx-3.0-gtk3 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=0 -fno-var-tracking-assignments -I../../../../common -I../../../../../common  -c -o test.o test.cpp
[... cut ...]
g++ -o test test.o vscpl2drv-sim.o simulation.o vscpremotetcpif.o vscphelper.o guid.o crc.o crc8.o vscp_aes.o vscpmd5.o mongoose.o fastpbkdf2.o vscpbase64.o  -L@top_wxbuilddir@lib  -lssl -lz -lrt -lm -lcrypto -lpthread -ldl   -L/usr/lib64 -pthread   -lwx_baseu-3.0-gtk3 -lwx_baseu_xml-3.0-gtk3 -lwx_baseu_net-3.0-gtk3 
mkdir -p /srv/vscp/drivers/level2
/usr/bin/install -c vscpl2drv-sim.so /srv/vscp/drivers/level2
make[1]: Leaving directory '/home/vscp/vscp/src/vscp/drivers/level2/sim/linux'
/usr/bin/install -c -b -m755 config_examples/vscpd.startup_script_debian /etc/init.d/vscpd
update-rc.d vscpd defaults
make: update-rc.d: Command not found
make: *** [Makefile:45: install] Error 127

grodansparadis, 2018-06-11:
I see there is an error when vscpd is built I will check on that
tomorrow.  Else it looks ok except for the installation of the startup
script.  Is this ArchLinux?

On 2018-06-11 21:51, Jean-Marc Beaune wrote:
> In /usr/local/bin.
>
> Here is the output of 'sudo make install':
>
> $ sudo make install
> hostname: invalid option -- 'I'
> Usage: hostname [-v] {hostname|-F file} set hostname (from file)
> domainname [-v] [...]

mamac, 2018-06-12:
It's Gentoo Linux.
In the meantime I am updating the whole operating system and I have a problem updating glibc.
It might be worth waiting for the OS to be fully updated and try again...

grodansparadis, 2018-06-12:
I tested the head code here and can confirm there is a problem with the
compilation of vscpd and the sim driver. I am busy with the
documentation move at the moment but will merge the development branch
to the head next week and fix and this problem.

Use the release code in the meanwhile. It should work except for the
startup script. Need to find a way to install it on non lsb systems.

mamac, 2018-06-14:
I am investigating the warning below:

> cc1plus: warning: /usr/local/include: not a directory

It is actually interpreted as an error and I can't update the system due to that.

Indeed, /usr/local/include is a file and it seems to be created by vscp build:

https://paste.pound-python.org/show/rTFEFpf8dUKVXm6PxATb/

I will delete this file and create a directory instead gcc can work again.

grodansparadis, 2018-06-14:
lsb systems should have /usr/local/include and configure try to point
install stuff there. The helper lib have these lines i

bindir = ${exec_prefix}/bin
libdir = ${exec_prefix}/lib
datadir = ${prefix}/share
includedir = ${prefix}/include

in the install.

which will be

$(INSTALL_PROGRAM) ../vscphelperlib.h $(includedir)

in the install section giving the result you get.

Solution is to use another prefix in configure (--help) or create the
dir beforehand. It is strange it is't there really. wxWidgets usually go
there to.

grodansparadis, 2018-09-26:
> Hi Ake,
> I just pulled the latest from master and compiled but still vspworks
> is not created.
> I can't see any error upon compilation thought...

In this

           VSCP configuration
--------------------------------------------
VSCP version          : 13.1.1.4 Aluminium (13.1.1)
System                : Linux fluorine 4.15.0-25-generic #27-Ubuntu SMP
Mon Jul 2 08:35:25 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
Architecture          : x86_64
VSCP GUI Build        : yes
VSCP install dir      : /srv/vscp
build flavour         : debug
linking for wxWidgets : shared
Raspberry Pi build    : yes

after ./configure  do you have

VSCP GUI Build        : yes

/Ake

grodansparadis, 2018-09-26:
What is your output from "wx-config --libs"

mamac, 2018-09-26:
ha, it says no to the gui:

 VSCP configuration
--------------------------------------------
VSCP version          : 13.1.1.4 Aluminium (13.1.1)
System                : Linux gentoo-jm 4.12.5-gentoo #1 SMP Tue Aug 22 23:23:10 CEST 2017 x86_64 Intel(R) Core(TM) i5-6300U CPU @ 2.40GHz GenuineIntel GNU/Linux 
Architecture          : x86_64
VSCP GUI Build        : no
VSCP install dir      : /srv/vscp
build flavour         : release 
linking for wxWidgets : shared 
Raspberry Pi build    : no

mamac, 2018-09-26:
her is the output of wx-config --libs :
-L/usr/lib64 -pthread   -lwx_gtk3u_xrc-3.0-gtk3 -lwx_gtk3u_html-3.0-gtk3 -lwx_gtk3u_qa-3.0-gtk3 -lwx_gtk3u_adv-3.0-gtk3 -lwx_gtk3u_core-3.0-gtk3 -lwx_baseu_xml-3.0-gtk3 -lwx_baseu_net-3.0-gtk3 -lwx_baseu-3.0-gtk3

grodansparadis, 2018-09-26:
OK I found the problem. The configure file check for gtk2 only.

Two solutions at the moment for this

1.) use gtk2

2.) Edit the configure.ac file in the root folder. Change "wx_gtk2*"  to
"wx_gtk3*" on line 213 and "wx_gtk2"  to "wx_gtk3" 214 and after that
issue autoconf as root and then run ./configure

As an alternative add

AC_MSG_CHECKING([for wx_gtk3* wxWidgets libs])
if sh -c "$WX_CONFIG_WITH_ARGS --basename" 2>/dev/null | grep wx_gtk3
>/dev/null; then
    dnl wxGTK is installed, build the GUI applications also
    PROJ_SUBDIRS="$PROJ_SUBDIRS $PROJ_SUBDIRS_GUI"
    vscp_gui=yes
    AC_MSG_RESULT([Found; building GUI apps])
fi

after the test block for gtk2.

/Ake

grodansparadis, 2018-09-26:
Even better change line 214 to

if sh -c "$WX_CONFIG_WITH_ARGS --basename" 2>/dev/null | grep -e wx_gtk2 -e wx_gtk3 >/dev/null; then

and then do "autoconf" as root.

/Ake

commit b1ba8dad (2018-09-27): Fixed tcp/ip interface chid bug. Fixes  #243
closed 2018-09-27 by grodansparadis