grodansparadis/vscp #247: tcpip interface / ssl problem

BlueAndi, 2018-07-12:
The tcpip interface in the vscpd.conf is installed with

`
    <tcpip enable="true"
        interface="tcp://9598" />
`

which leads to an ssl init error `[TCP/IP srv thread] Failed to init. ssl.`

After changing it to
`    <tcpip enable="true"
        interface="tcp://9598"
        ssl_certificate="" />
`
the tcp/ip sever starts up without ssl.

grodansparadis, 2018-07-12:
OK noted. That is bad. Will change that so defaults works again.   Thanks.

BlueAndi, 2018-07-12:
But dont hurry. Good night!

closed 2018-07-13 by grodansparadis

grodansparadis, 2018-07-13:
The problem here was that the tcpip server certs was missing. 
"make_sample_certs" (or make install-system) install them now. Scripts
to generate them are in install_files/certs

Also note that the format for interfaces changed. "tcpip://" is removed
(should still work though). udp interface still use this format but will
also be changed.

commit d88da8aa (2018-07-13): Fixes #247