grodansparadis/vscp #251: Programming a new image via vscpWorks

BlueAndi, 2018-07-21:
As I promised, I will take a look to vscpWorks and the firmware update mechanism according the vscp algorithm. With the last bugfixes I am now able to. :-D

After the first fast look, I saw something what could be an issue for the users:
Because you always have to program a complete page on the mcu, the transfered block (page) must be complete. Otherwise the last block must be filled up with e.g. 0x00.
Therefore I would propose to extended vscpWorks here, that it automatically fills the rest of the last block up in case that the data size is lower than a page size.

I know that the users could do this by filling up in the .hex file by their own, but this is such a tech. low level detail, I fear it won't happen.

It is also possible to move the intelligence to the bootloader itself. That means a "program block" is sent independet of a received "block data acknowledge" and the bootloader recognizes that the page is not completly filled (but it contains at least one received data transfer), so it is filled up and programmed.
I would avoid this, because it changes the standard vscp algorithm, makes the bootloader more complex and its size will increase.

grodansparadis, 2018-07-21:
Hi Andreas,

for me the bootloader is for "any memory", "any technique". So what we
program can be RAM, EEPROM, Flash, just some logic entity,  file,
"configuration memory" or whatever else. The series of commands I came
up with was just a set that would work for all of the memory types I
could think of.  It is a memory located at some location. To implement
in most devices one have to through in some intelligence as a flash
programming, ram programming, eeprom programming must be logically layed
out in the address space.

So as I see it the only way to fulfill that abstraction is to put any
extra intelligence in the bootloader.

Also. A "user" uploads supplied firmware. A "maker" makes and/or adopts
the bootloader and supply bootloader files for users. It is possible
that "users" == "maker". But in either case "maker" has the
rsponsibility to supply a file that is in the correct format for a
specific device he/she made.

Another problem I have noticed with my unit's is that people can upload
wrong firmware to a device. This probably should be handled by the
algorithm as well.

grodansparadis, 2018-07-21:
>
> In my first check I focused on writing a new firmware to the flash,
> thats true.
> Independent of any kind of memory, all depends on the block size,
> which the node responds and the loaded data from the .hex file. The
> data size must always be a multiple of the block size, otherwise the
> standard vscp algorithm doesn't fit, isn't it?
yes that is true even if that does not have to be physical figures but
just logical.
>
> So the general question is, what to in this case independent of the
> kind of memory?
>
>   * After receiving the block size, abort in case its not a multiple
>     of it?
Yes I think that sounds as the right hing to do.
>
>   * Provide a checkbox for filling the block up? I wouldn't do this
>     always, but only on request by the user!
Yes why not.
>
> And I confirm that programming of something invalid is a case! ;-)
> My bootloader supports here a minimal application, where you should be
> able to reprogram it in any case.
OK
>
> Whats about the forest fire at the moment? Is it stopped right now so
> it doesn't grow anymore or not? Hopefully the first one!
Still out of control. Not possible to put down, just restrict. But still
growing. But not in our direction at the moment thankfully. The smoke is
heavy here from time to time. Thanks for your helicopter support
(Germany) btw.  :) Forty fire vehicles with men from Poland came today.
The area here is much like a war zone at the moment.

BlueAndi, 2018-07-21:
> Still out of control. Not possible to put down, just restrict. But still growing. But not in our direction at the moment thankfully. The smoke is heavy here from time to time. Thanks for your [...]

Sounds not good. I cross the fingers and hope they are able to stop it in the next days. I organized and executed a fire fighting training in the forest here this year. But this was just nothing compared to the real situation at your side right now. I have a lot of respect for all which are fighting against and also for the ones supporting them.

Let it rain!

BlueAndi, 2018-07-22:
The following is just for my understanding:
* The "enter bootloader mode ACK" event returns the block size and max. number of blocks, which are only related to flash memory. Right?
* The "start block data transfer" event selects the kind of memory, but may have a different block size and a different number of max. blocks.

How to deal with that?

grodansparadis, 2018-07-22:
On 2018-07-22 17:35, Andreas Merkle wrote:
>
> The following is just for my understanding:
>
>   * The "enter bootloader mode ACK" event returns the block size and
>     max. number of blocks, which are only related to flash memory. Right?
>
yes

>   * The "start block data transfer" event selects the kind of memory,
>     but may have a different block size and a different number of max.
>     blocks.
>
No. Block size must be abstracted. You present a memory space of a
number of blocks each with a block size. This can be any kind of
storage. Normally that should be enough.  Optionally specify type, image
but normally no need.

So as a maker.
 - Specify blocksize.
 - Specify # blocks. 
 - Lay out your memory among the different blocks.
 - When a block is about to be written. Init it as needed by the memory
type.
 - When written (programmed) write and verify. Disregard excess data,
use default for non written data.
- etc

-- 
Stay Hungry - Stay Foolish!
PGP: EF7161865D3E1CCE61E2F43464B570FEC2BE58AA
Åke Hedman
Brattbergavägen 17, 82050 LOS, SWEDEN
http://www.akehedman.se http://www.vscp.org http://www.grodansparadis.com

BlueAndi, 2018-07-23:
Example:
[image]

Notes:
* The block number is the absolute number in the abstract block space.
* The bootloader derives from the block number the
  * memory type
  * address in memory
  * block size
* The vscpWorks address layout is specific for the standard vscp algorithm.

All correct? Sorry, but I want just to be sure before changing anything. :-)

grodansparadis, 2018-07-26:
> grafik
>       o memory type
>       o address in memory

I would set one blocksize for all. Thats the abstraction. So here 128.
It's better to use registers to abstract single pos eeprom, ram etc
locations.  For the bootloader algorithm the intended use is to init.
all of it. This can obviously be a problem as the blocksize may not be
an even multiple  of eeprom/ram etc. But that can be handle locally. 

For example

A 192 byte EEPROM (a bit odd but anyway) with a block size of 128

- First block 0-127

- Second block 128-191

Throw away excess data in second transfer if any.  Calc. checksum on
programmed content.

BlueAndi, 2018-07-26:
Ok, lets keep the block size for the 2nd and 3rd memory type.
But at least for the 3rd memory type vscpWorks wouldn't know where to start with the block number, because the max. number of blocks for the 2nd memory type are missing.

Hmm ...

grodansparadis, 2018-07-26:
But you the designer know that. And you the designer make the data file.
So you put out a number of blocks and map them to memory types. The
optional fields just complicates things in most cases but can be used if
no other way is possible.

There are other ways to that can be used to load data and I would prefer
such a method if I want to know the block size. Use a register on the
device to select which memory to load data to. Then you have the total
blocks and the blocksize different for each memory type if you want.
This is also a better way to use than the optional fields for flash
image areas.

I use registers to write a multibyte key to unlock flash write and
special register write for a device also for security. Registers are
there and are great. ;-)

Anyhow the bootloader is intended for flash in the first place and
probably will be used for that in almost all cases. Initating eeprom and
other memories have better methods IMHO

Also remember the crc should be calculated over all blocks. Mixing makes
things hard if you don't write data everywhere.

BlueAndi, 2018-07-26:
> Anyhow the bootloader is intended for flash in the first place and
probably will be used for that in almost all cases. Initating eeprom and
other memories have better methods IMHO

Maybe my approach was wrong. I tried to figure out how it should workt by checking the events in the specification and how its "partly" implemented in vscpWorks. Could it be that the parts for config and EEPROM data programming in vscpWorks was just a try, but never finished/used? If yes, then this was misleading me ...

So I would remove the obsolete parts in vscpWorks and add just the checkbox for filling the last block up. Thats it. :-)

grodansparadis, 2018-07-26:
Dinesh did something in the past but as far as I could see of the code
(I think you pointed this out at the time) it is not working code and
has never been even if it was stated too be at the time.

Can safely be removed AFAIK.

BlueAndi, 2018-07-31:
I removed the obsolete parts and tried to add a checkbox for filling up the last block.
Unfortunately I failed adding the checkbox. It seems the whole UI was created with a tool.
Anyway I simulated successful a firmware update with my pc node.

grodansparadis, 2018-08-01:
It (dialogs) was autogenerated many years ago but is no longer so it is
safe to edit the code. I try to change variable names to sensible once
as I go into different parts.

Nice to here this btw.

grodansparadis, 2019-03-25:
VSCP Works is deprecated

closed 2019-03-25 by grodansparadis