grodansparadis/vscp #254: VSCP 13.1.1.2 daemon exception regarding UTF-8

BlueAndi, 2018-09-22:
Connecting via telnet session to the daemon results in

```
2018-09-22T21:07:34 - [TCP/IP srv] -- Connection accept.
/usr/include/wx-3.0/wx/string.h(1748): assert "!utf8 || !*utf8 || !s.empty()" failed in FromUTF8Unchecked(): string must be valid UTF-8 [in thread 999ff450]

Call stack:
[00] 0x52e616
[01] wxThread::CallEntry()
[02] 0xb6c33b98
```

Target is a beaglebone black board.

I traced the problem till
tcpipsrv.cpp @ line 218, outputs "[TCP/IP srv] -- Connection accept.\n"

Additional I am confused about using the _("") macro, which is used for translation and different from _T("") … could it be that a lot of the strings should use _T(""), instead of _("") for unicode support?

grodansparadis, 2018-09-24:
Strange I can't see that on the other platforms.  What version of wxWidgets is it?

for _() it is OK, just a bit slower than _T()

BlueAndi, 2018-09-24:
I am using wxWidgets v3.0.2

BlueAndi, 2018-09-24:
Actually I try to find out why the sending a websocket event to all interfaces (GUID = "-") is not sent to the CAN bus via daemon. During that I stumpled over this. ;-)

grodansparadis, 2018-09-24:
Hmm... same here.  But no Beaglebone to test on. But seven Pi's and some
more.

grodansparadis, 2018-09-25:
btw to send events on a specific level I interface or rather to a
specific level I node on a specific level I interface use

class + 512  for class.

16-first databytes is GUID for level I interface with node id in the two
LSB bytes (14/15)

BlueAndi, 2018-09-25:
I tried using the interface guid without node id, but does not worked.
Hope to get a little bit time tomorrow for further investigation.

grodansparadis, 2019-03-25:
Open if still a problem

closed 2019-03-25 by grodansparadis