grodansparadis/vscp #268: Remove your political support for Ukraine

gitgit3, 2022-03-24:
Your support for Ukraine and your anti-Putin rhetoric  is certainly an issue with your project so I thought Id report it as such.

You need to be very careful that you're not backing the wrong horse. I am sure that you mean well but you have been drawn in completely by Western propaganda. I actually lived and worked in Ukraine before the Maiden protests and subsequent coup took place. I can tell you for sure that western NGO's were paying students to protest. After the coup, Poroshenko was installed by the West and immediately outlawed the Russian language and started to persecute all Russians. For 8 long years now, the Kiev government has been bombing and killing civilians in Donbas. The Ukrainians are terrified by their military which are filled with Nazis and regularly abuse civilians, particularly the Russian speaking. In fact, many Ukrainians are relieved that Russia has intervened. Maybe there might have been another way? Maybe Putin could have done more to stop the West flooding Ukraine with military equipment? Maybe Putin could have done something else to stop the build up of far right anti-Russian military groups in Ukraine? I can't see how to be honest. None of us want war except, it seems, our Western leaders. But whatever your point of view on Ukraine, a software development site is not the place voice those opinions.

Please remove this ill-founded anti-Russian rhetoric and get back to what you are clearly very good at. You are clearly an expert at software development but equally, you are no expert at international politics.

rtsn, 2022-03-24:
Please close this non-issue. 

Russian war machine, go fuck yourself.

grodansparadis, 2022-03-24:
No comments are needed for fools' talk.   We support Ukraina and the free world all the way, every minute, until death.

grodansparadis, 2022-03-24:
And yes, go fuck yourself!!!

closed 2022-03-24 by grodansparadis

gitgit3, 2022-03-24:
Your rudeness shows you are clearly a signed up member of Azov. Death to Azov and all its members.

gedass2000, 2022-03-24:
Слава Україні! Героям слава! Putin khuylo!

dinguluer, 2022-03-25:
Don't fight friends ... for agenda of people sitting in power  ...!!

Russia wants to kick NATO out of its neighbourhood & *ROTHSCHILD banking
family* led (*USA & NATO*) want to control the resources of every country.

We are common citizens who will die in war as soldiers.. rich people who
want war will sit in comfort zone in there home...!!